stm32duino/STM32FreeRTOS #3: fatal error: stm32_def.h

heathkit, 2018-01-24:


fpistm, 2018-01-24:
Hi,
unfortunately, this library is designed to be used with this core:
https://github.com/stm32duino/Arduino_Core_STM32
You use the one from @rogerclarkmelbourne:
https://github.com/rogerclarkmelbourne/Arduino_STM32
Which are not the same.
I know there is a request in platform.io community to support this core:
https://community.platformio.org/t/stm32-nucleo-discovery-boards-for-arduino/3027

Note that there is also a port of FreeRTOS for @rogerclarkmelbourne  core:
https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F1/libraries/FreeRTOS900

rogerclarkmelbourne, 2018-01-24:
There is a several version of FreeRTOS bundled as a library in https://github.com/rogerclarkmelbourne/Arduino_STM32, so I"m not sure why @heathkit would try to use this library

See

https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F1/libraries/FreeRTOS900

https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F1/libraries/FreeRTOS821

https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F1/libraries/FreeRTOS

fpistm, 2018-01-24:
@rogerclarkmelbourne probably because this one has just been referenced by Arduino in the Library manager. ;)

heathkit, 2018-01-24:
It was recently added to PlatformIO: http://platformio.org/lib/show/2093/STM32duino%20FreeRTOS

If you start a new STM32 project on PlatformIO and search for a FreeRTOS library, this is what you find. Is there a missing dependency?

rogerclarkmelbourne, 2018-01-24:
Umm

Platformio always throws up lots of problems :-(

The Platforino guys should really link to their primary sources and the forums for each one of them.

fpistm, 2018-01-24:
I don't know how platform IO handle dependencies.
All libraries related to this github organization are for arch stm32.
Those for @rogerclarkmelbourne core (based on leaflabs) are for arch stm32f1 or stm32f4.
https://github.com/stm32duino/STM32FreeRTOS/blob/24cd55d0fbcf93ca742e6f6a0ece3a41e8c153b3/library.properties#L9

fpistm, 2018-01-24:
@rogerclarkmelbourne it seems platform io automatically import new library published in Ardhuino library manager.

rogerclarkmelbourne, 2018-01-24:
@fpistm 

This is a platformio issue, if they don't import the right libraries to match the core.

I've never been contacted by the Platformio team.
I don't support Platformio in my core as its too much work. (remember this is not my day job)
It says explicitly in my Readme.md that its for the Arduino IDE only.

fpistm, 2018-01-24:
I know ;)
I've only answered to @heathkit  about why this lib do not work as it is not for the same core.
@heathkit I close this issue as unrelated to this library.

fpistm, 2018-01-25:
In fact, there is at least an issue with the lib. It should not use internal core  header and use standard Arduino.h as entry point.

closed 2018-01-30 by fpistm

mfatihkoc, 2019-04-17:
Hey, I have these error on my code, what to do? could you please help me?

BalancingWii:18:17: error: def.h: No such file or directory

compilation terminated

exit status 1
def.h: No such file or directory

fpistm, 2019-04-17:
Hi @heathkit 
if you used PIO then you have to enter an issue on its related GitHub.
Moreover, just telling it doesn't compile is not enough.
Which IDE, core version, sketch example, all relevant and helpful info,...

PR #4 (2018-01-30): Avoid use internal core header file use Arduino.h instead
Up to the used core to define properly what the library need.
Mainly the _cortex M_ core to use:
Ex: for **STM32F103** based board:
`#define __CORTEX_M (0x03U) /*!< Cortex-M Core */`

https://github.com/stm32duino/Arduino_Core_STM32 uses the CMSIS which defined it.
https://github.com/ARM-software/CMSIS/blob/6891719834ac6ca2a79e8ded00620faffae10ae8/CMSIS/Include/core_cm3.h#L79

Fix #3
commit d1d80bb2 (2018-01-30): Avoid use internal core header file use Arduino.h instead

Up to the used core to define properly what the library need.
Mainly the cortex M core to use:
Ex: for STM32F103 based board:
#define __CORTEX_M (0x03U) /*!< Cortex-M Core */

https://github.com/stm32duino/Arduino_Core_STM32 uses the CMSIS which defined it.
https://github.com/ARM-software/CMSIS/blob/6891719834ac6ca2a79e8ded00620faffae10ae8/CMSIS/Include/core_cm3.h#L79

Fix #3