stm32duino/STM32FreeRTOS #13: xTaskGetTickCount and uxTaskPriorityGet don't work

nopnop2002, 2019-03-21:
code:

	#include <STM32FreeRTOS.h>

	void setup() {
	  Serial.begin(115200);
	  while(!Serial);
	  Serial.println("Serial Start");
	}

	void loop() {
	  static int counter=0;
	  TickType_t nowTick;
	  UBaseType_t Priority;
	  
	  Serial.print("counter="); 
	  Serial.println(counter);

	  nowTick = xTaskGetTickCount();
	  Serial.print("nowTick=");
	  Serial.println(nowTick);

	  Priority = uxTaskPriorityGet(NULL);
	  Serial.print("Priority=");
	  Serial.println(Priority);
	  
	  counter++;
	  delay(1000);
	}

result:

	nowTick=0
	Priority=134218401
	Serial counter=101
	nowTick=0
	Priority=134218401
	Serial counter=102
	nowTick=0
	Priority=134218401
	Serial counter=103
	nowTick=0
	Priority=134218401
	Serial counter=104

fpistm, 2019-03-21:
This seems logical as you do not start the scheduler:
The tick count is the total number of tick interrupts that have occurred **since the scheduler was started**. `xTaskGetTickCount()` returns the current tick count value.

```
#include <STM32FreeRTOS.h>

static int counter = 0;

// Declare a semaphore handle.
SemaphoreHandle_t sem;
//------------------------------------------------------------------------------
/*
   Thread 1, turn the LED off when signalled by thread 2.
*/
// Declare the thread function for thread 1.
static void Thread1(void* arg) {
  UNUSED(arg);
  while (1) {
    Serial.print("Thread1: counter=");
    Serial.println(counter++);
    
    Serial.print("nowTick=");
    Serial.println(xTaskGetTickCount());

    Serial.print("Priority=");
    Serial.println(uxTaskPriorityGet(NULL));
    // Wait for signal from thread 2.
    xSemaphoreTake(sem, portMAX_DELAY);
  }
}
//------------------------------------------------------------------------------
/*
   Thread 2, turn the LED on and signal thread 1 to turn the LED off.
*/
// Declare the thread function for thread 2.
static void Thread2(void* arg) {
  UNUSED(arg);
  while (1) {
    Serial.print("Thread2: counter=");
    Serial.println(counter++);

    Serial.print("nowTick=");
    Serial.println(xTaskGetTickCount());

    Serial.print("Priority=");
    Serial.println(uxTaskPriorityGet(NULL));

    // Sleep for 1000 milliseconds.
    vTaskDelay((1000L * configTICK_RATE_HZ) / 1000L);

    // Signal thread 1 to turn LED off.
    xSemaphoreGive(sem);

    // Sleep for 200 milliseconds.
    vTaskDelay((1000L * configTICK_RATE_HZ) / 1000L);
  }
}
//------------------------------------------------------------------------------
void setup() {
  portBASE_TYPE s1, s2;

  Serial.begin(115200);

  // initialize semaphore
  sem = xSemaphoreCreateCounting(1, 0);

  // create task at priority two
  s1 = xTaskCreate(Thread1, NULL, configMINIMAL_STACK_SIZE, NULL, 2, NULL);

  // create task at priority one
  s2 = xTaskCreate(Thread2, NULL, configMINIMAL_STACK_SIZE, NULL, 1, NULL);

  // check for creation errors
  if (sem == NULL || s1 != pdPASS || s2 != pdPASS ) {
    Serial.println(F("Creation problem"));
    while (1);
  }

  // start scheduler
  vTaskStartScheduler();
  Serial.println("Insufficient RAM");
  while (1);
}

//------------------------------------------------------------------------------
// WARNING idle loop has a very small stack (configMINIMAL_STACK_SIZE)
// loop must never block
void loop() {
  // Not used.
}
```

```
Thread2: counter=5
nowTick=4002
Priority=1
Thread1: counter=6
nowTick=5002
Priority=2
Thread2: counter=7
nowTick=6002
Priority=1
Thread1: counter=8
nowTick=7002
Priority=2
Thread2: counter=9
nowTick=8002
Priority=1
Thread1: counter=10
nowTick=9002
Priority=2
```

nopnop2002, 2019-03-21:
I add vTaskStartScheduler() to my sketch.

But pcTaskGetName(NULL) & uxTaskPriorityGe(NULL) is not correct.
It get about IDEL task information.

code:

	#include <STM32FreeRTOS.h>

	void setup() {
	  Serial.begin(115200);
	  while(!Serial);
	  Serial.println("Serial Start");
	  vTaskStartScheduler();
	//  Serial.println("Insufficient RAM");
	}

	void loop() {
	  static int counter=0;
	  TickType_t nowTick;
	  UBaseType_t Priority;

	  Serial.println();
	  Serial.print("counter="); 
	  Serial.println(counter);

	  nowTick = xTaskGetTickCount();
	  Serial.print("nowTick=");
	  Serial.println(nowTick);

	  Priority = uxTaskPriorityGet(NULL);
	  Serial.print("Priority[null]=");
	  Serial.println(Priority);

	  Serial.print("TaskName[null]=");
	  Serial.println(pcTaskGetName(NULL));

	  Priority = uxTaskPriorityGet(0);
	  Serial.print("Priority[0]=");
	  Serial.println(Priority);

	  Serial.print("TaskName[0]=");
	  Serial.println(pcTaskGetName(0));

	  counter++;
	  delay(1000);
	}

result:

	counter=64
	nowTick=64193
	Priority[null]=0
	TaskName[null]=IDLE
	Priority[0]=0
	TaskName[0]=IDLE

fpistm, 2019-03-21:
Well, your code doesn't rely on FreeRTOS. You do not start any task...
Please see the examples and the FreeRTOS API.
Example for vTaskStartScheduler(): https://www.freertos.org/a00132.html
Arduino loop is in fact never used as this is not a task.

nopnop2002, 2019-03-21:
For example the following cord is added automatically in ESP32.

	#include "freertos/FreeRTOS.h"
	#include "freertos/task.h"
	#include "esp_task_wdt.h"
	#include "Arduino.h"

	TaskHandle_t loopTaskHandle = NULL;

	#if CONFIG_AUTOSTART_ARDUINO

	#if CONFIG_FREERTOS_UNICORE
	#define ARDUINO_RUNNING_CORE 0
	#else
	#define ARDUINO_RUNNING_CORE 1
	#endif

	bool loopTaskWDTEnabled;

	void loopTask(void *pvParameters)
	{
	    setup();
	    for(;;) {
	        if(loopTaskWDTEnabled){
	            esp_task_wdt_reset();
	        }
	        loop();
	    }
	}

	extern "C" void app_main()
	{
	    loopTaskWDTEnabled = false;
	    initArduino();
	    xTaskCreatePinnedToCore(loopTask, "loopTask", 8192, NULL, 1, &loopTaskHandle, ARDUINO_RUNNING_CORE);
	}

	#endif

How are setup() and loop() handled in AVR?

fpistm, 2019-03-21:
Honestly, I don't know how this is handle in AVR.
This is not the case for this core. If someone would try enable it, PR can be submit on the core and library.

nopnop2002, 2019-03-21:
I understood that function in the loop() can't be used under AVR environment..

Thank you so mach.

fpistm, 2019-03-21:
Setup is call only once.

Loop is called thanks:
https://github.com/stm32duino/STM32FreeRTOS/blob/8e6689613649bab2d4dd8a140513cdcee3b40892/src/STM32FreeRTOS.c#L73-L87

But I do not advise to use it
It's up to end user to manage properly their tasks. FreeRTOS allows several tuning/configuration.

nopnop2002, 2019-03-21:
I understand that loop() is called from vApplicationIdleHook().
I understand that loop() is not task.
Thank you very much.
Please close this issues.

fpistm, 2019-03-21:
Setup is called like for other Arduino example at the beginning. It switch to rtos when scheduler is called. That's all.

nopnop2002, 2019-03-21:
Thank you.

Please close this issues.

closed 2019-03-21 by fpistm