stm32duino/STM32FreeRTOS #16: HID will not work if STM32FreeRTOS.h is included.

ogatatsu, 2019-04-13:
env: Arduino 1.8.9 + Arduino_Core_STM32 + RobotDyn STM32 MINI
code:
```c++
// It does not work if there is this line.
// #include <STM32FreeRTOS.h>
#include "Keyboard.h"

void setup()
{
  Keyboard.begin();
  delay(1000);
  Keyboard.print("test");
  while (1);
}

void loop()
{
}
```

fpistm, 2019-04-13:
Hi @ogatatsu 
I will check that.

fpistm, 2019-04-15:
Hi @ogatatsu 
It is linked to the heap management using newlib.
Using the `configMEMMANG_HEAP_NB             3` and it works.
https://github.com/stm32duino/STM32FreeRTOS/blob/master/src/FreeRTOSConfig_Default.h#L92
Unfortunately, I will not be able to help on this, it was provided as several user request it.
Anyway, it's up to end user to configure properly its usage.

ogatatsu, 2019-04-16:
Thank you very much. I was able to work it.

closed 2019-04-18 by fpistm

Yannick-Marietti, 2019-05-08:
Hello

Thank you very much for this reporting !

I had similar issues,  I applied `configMEMMANG_HEAP_NB 3` correction following your comments, and everything works fine now.

To be more precise, newlib had adverse effects on Arduino Interrupts (not working) and also on the initialization code of the SPBTLE-RF module on the Discovery IoT Node board.