stm32duino/STM32FreeRTOS #17: vTaskDelay Don't seem to be working

SocketNet, 2019-04-16:
#include <Arduino.h>
#include <STM32FreeRTOS.h>

#define LED PC13

void task(void *pvParameters)
{
  Serial.println("task start");
  for (;;)
  {

    digitalWrite(LED, HIGH);
    vTaskDelay(100);
    digitalWrite(LED, LOW);
    vTaskDelay(100);
  }
}

void setup()
{
  Serial.begin(115200);
  Serial.println("Serial start");
  pinMode(LED, OUTPUT);
  xTaskCreate(task, "task", 128, NULL, 2, NULL);
  vTaskStartScheduler();
}

void loop()
{
  for (int i = 65; i < 91; i++)
  {
    Serial.println((char)i);
    delay(1000);
  }
}

fpistm, 2019-04-16:
Loop is called thanks:
https://github.com/stm32duino/STM32FreeRTOS/blob/8e6689613649bab2d4dd8a140513cdcee3b40892/src/STM32FreeRTOS.c#L73-L87

Your loop take too much time. You should not called delay. In a general way, avoid to use loop, create dedicated task. Moreover, your blink is very quick ( 100 ms)

SocketNet, 2019-04-17:
thank you but i try you frBlinkPrint.ino example  into stm32f103c8t6 led not work

fpistm, 2019-04-17:
@SocketNet 
I've just tried with my BP and it's works fine.

SocketNet, 2019-04-18:
#### myconfig

```
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
upload_protocol = serial
board_build.core = STM32
```

#### code 
```
  #include <Arduino.h>
  #include <STM32FreeRTOS.h>

  void task(void *pvParameters)
  {
    Serial.println("task start");
    for (;;)
    {
      for (int i = 65; i < 91; i++)
      {
        Serial.println((char)i);
        vTaskDelay(500 / portTICK_RATE_MS);
      }
    
    }
  }

  void setup()
  {
    Serial.begin(9600);
    Serial.println("Serial start");
    xTaskCreate(task, "task", configMINIMAL_STACK_SIZE + 128, NULL, tskIDLE_PRIORITY + 2, NULL);
    vTaskStartScheduler();
    while(1);
  }

  void loop()
  {
    
  }
```
The console only outputs  "A", and the task seems to stop as soon as it arrives vTaskDelay and never continues.

fpistm, 2019-04-18:
OK, you used PIO.
So, you should open an issue on its GitHub.
It works with Arduino IDE and we don't support PIO.

SocketNet, 2019-04-18:
ok Thanks again

closed 2019-04-18 by SocketNet

ivankravets, 2019-05-09:
@fpistm Please merge this https://github.com/stm32duino/STM32FreeRTOS/pull/18 and we will reindex this library in our registry. We need to instruct PlatformIO Build System to not archive library and link as plain objects.