stm32duino/STM32FreeRTOS #25: Hardfault on CMSIS OS v2

danyhm, 2020-05-03:
Hello,

I installed the library via Arduino library manager and tried running the examples on an STM32 blue pill.

all examples except the ones in CMSIS-RTOSv2 run fine. if I try to run the examples in CMSIS-RTOSv2 the MCU enters a hardfault with 3 (or 4) blinks immediately.

I narrowed the problem down to be from the CMSIS-RTOSv2 configurations because whenever I enable configUSE_CMSIS_RTOS_V2 the hardfault happens otherwise it runs fine.

I hope it helps.

fpistm, 2020-05-04:
Hi @danyhm,
I've tested and you're right, anyway as mentioned in the [README.md](https://github.com/stm32duino/STM32FreeRTOS#limitations):

> No CMSIS-RTOSv2 support provided. It is provided as example.

Do not hesitate to make a PR to fix this, it is a community organization and all contributions are welcome 😉

danyhm, 2020-05-04:
@fpistm  I'll definitely look into it once my current project is done!

fpistm, 2021-04-16:
@danyhm 
just for my interest which target you used ?

danyhm, 2021-04-16:
I used the bluepill with CMSIS-RTOS-V1

EasyNetDev, 2021-08-15:
Hi,

I've notice the same behavior with my BlackPill STM32F411CE. I tried CMSIS-RTOSv2 with a very-very simple task to blink the builtin LED at 1second and after uploading the software in the MCU it blinks fast 3 times and it stays on for a while then it repeats the sequence.
Is this CMSIS-RTOSv2 supported on F411CE or only on F411RE? They have same flash size, RAM size and speed.

Kind regards.

fpistm, 2021-11-17:
Close as no update since a while and as:
> No CMSIS-RTOSv2 support provided. It is provided as example.

closed 2021-11-17 by fpistm