stm32duino/STM32FreeRTOS #26: vApplicationGetIdleTaskMemory not present if static allocation is on

danyhm, 2020-05-03:
Hello,

the vApplicationGetIdleTaskMemory function is not present if configSUPPORT_STATIC_ALLOCATION is 1. It seems it needs to be supplied manually. from CubeMX we have:
```
/* GetIdleTaskMemory prototype (linked to static allocation support) */
void vApplicationGetIdleTaskMemory( StaticTask_t **ppxIdleTaskTCBBuffer, StackType_t **ppxIdleTaskStackBuffer, uint32_t *pulIdleTaskStackSize );

static StaticTask_t xIdleTaskTCBBuffer;
static StackType_t xIdleStack[configMINIMAL_STACK_SIZE];
  
void vApplicationGetIdleTaskMemory( StaticTask_t **ppxIdleTaskTCBBuffer, StackType_t **ppxIdleTaskStackBuffer, uint32_t *pulIdleTaskStackSize )
{
  *ppxIdleTaskTCBBuffer = &xIdleTaskTCBBuffer;
  *ppxIdleTaskStackBuffer = &xIdleStack[0];
  *pulIdleTaskStackSize = configMINIMAL_STACK_SIZE;
}
````

which needs to be supplied in the sketch.

maybe update the docs ? or provide examples for configSUPPORT_STATIC_ALLOCATION

fpistm, 2020-05-04:
Hi @danyhm 
This library is provided as a base to use FreeRTOS with the STM32 core not all configurations are provided nor supported.
User can customize the configuration but it is up to end user to provide/implement the required code.
I guess you enabled the `configUSE_CMSIS_RTOS_V2` then  `configSUPPORT_STATIC_ALLOCATION ` is enabled. 
Even in this case this is the user who need to provide it as described in the documentation:

https://www.freertos.org/a00110.html#configSUPPORT_STATIC_ALLOCATION

> If configSUPPORT_STATIC_ALLOCATION is set to 1 then the application writer must also provide two callback functions: vApplicationGetIdleTaskMemory() to provide the memory for use by the RTOS Idle task, and (if configUSE_TIMERS is set to 1) vApplicationGetTimerTaskMemory() to provide memory for use by the RTOS Daemon/Timer Service task. 

This is the case for several configurations and the library could not provide them to let user properly implements his solution properly.

danyhm, 2020-05-04:
@fpistm no actually i disabled CMSIS V2 because of the hardfault. then manually enabled static allocation and then the compiler gave me the linkage error. I looked into it using CubeMX generated code and found out that It has to be implemented manually which I did.

fpistm, 2020-05-04:
OK. So you do what was required ;)
I close this issue has there is no problem. Code have to be added/handled at application level.

closed 2020-05-04 by fpistm

wystewart, 2024-04-03:
just want to add that if your sketch file is cpp, then you have to add  extern "C"  in front otherwise linker won't find it..  Would be nice if this was on a wiki page.