stm32duino/STM32FreeRTOS #29: SerialUSB and FreeRTOS setup/startup order example

atrips, 2020-05-12:
Hello,

I'm trying to use SerialUSB w/ FreeRTOS but I have issue to start tasks before SerialUSB initialized.

Let's consider example code: https://github.com/stm32duino/STM32FreeRTOS/blob/master/examples/frBlinkPrint/frBlinkPrint.ino
Problem is that `while(!Serial) {}` blocking till ACMx device not connected. I tried to setup serial from task but without success. ACM device seems to not initialized right and not visible from OS.

Thanks.

fpistm, 2020-05-12:
Hi @atrips 
you can try this;
https://github.com/stm32duino/STM32FreeRTOS/issues/16#issuecomment-483173239

anyway, I will not able to help on this.

fpistm, 2020-05-19:
Hi @atrips 
I've tested the frBlinkPrint.ino and SerialUSB and did not have any issue with USB connected.
If your issue is that it's block to `while(!Serial);` then remove it. It is to ensure the Serial is up and running which is a good idea, anyway up to you do your own implementation and FreeRTOS config.
I've tested with default memman and then without `while(!Serial);' it works until I connect the USB.
But defining `#define configMEMMANG_HEAP_NB 3` in  STM32FreeRTOSConfig_extra.h works fine when I plugged the USB later as mentioned in #16.

So, I close it.

closed 2020-05-19 by fpistm

atrips, 2020-05-19:
Hello @fpistm ,

Thank you for the time you waste investigate the issue. I did not reply because I was not sure make my tests right. So...

Having  `while(!Serial);  and #define configMEMMANG_HEAP_NB 3` gives me proper OS level USB initialization but tasks does not starts till ACM device connected (from some app).

Commenting  `while(!Serial);` results:
```
[422541.924949] usb 3-5.4.4.4: new full-speed USB device number 19 using xhci_hcd
[422542.028102] usb 3-5.4.4.4: New USB device found, idVendor=0666, idProduct=5740, bcdDevice= 2.00
[422542.028106] usb 3-5.4.4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[422542.028109] usb 3-5.4.4.4: Product: BPILL_F103C8 CDC in FS Mode
[422542.028111] usb 3-5.4.4.4: Manufacturer: Atrips
[422542.028113] usb 3-5.4.4.4: SerialNumber: 326331693436
[422547.195618] usb 3-5.4.4.4: can't set config #1, error -110
```
NOTE:
I'm using platformio w/ following config:
```
...
build_flags =
    -fexceptions
    -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
    -D USBCON
;    -D USBD_USE_CDC
	-D HAL_PCD_MODULE_ENABLED
	-D USBD_VID=0x0666
	-D USBD_PID=0x0042
	-D USB_MANUFACTURER="\"Atrips\""
	-D USB_PRODUCT="\"BPILL_F103C8\""
...
```