stm32duino/STM32FreeRTOS #32: Library needs to include Arduino.h to compile OK

hreintke, 2020-10-20:
Hi,

I have a library that adds the option to use functional callbacks to the FreeRTOS timers.
That uses no Arduino functionality at all so I did not `include "Arduino.h"`
```
#include <functional>
#include "STM32FreeRTOS.h"
#include "timers.h"
```

That causes compile time errors :  error: template with C linkage
```
Compiling library "RTOSTicker"
"C:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\xpack-arm-none-eabi-gcc\\9.2.1-1.1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb "@C:\\Users\\Herman\\AppData\\Local\\Temp\\arduino_build_22616/sketch/build_opt.h" -c -Os -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -MMD "-IC:\\Users\\Herman\\Documents\\Arduino\\sketch_oct20b" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/avr" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/LL" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/usb" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/OpenAMP" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/usb/hid" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/usb/cdc" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/STM32F4xx_HAL_Driver/Inc" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/STM32F4xx_HAL_Driver/Src" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/STM32F4xx" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP/open-amp/lib/include" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP/libmetal/lib/include" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP/virtual_driver" -DSTM32F4xx -DARDUINO=10808 -DARDUINO_GENERIC_F411CE -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"GENERIC_F411CE\"" -DSTM32F411xE -DHAL_UART_MODULE_ENABLED "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\CMSIS\\5.5.1/CMSIS/Core/Include/" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/gcc/" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\CMSIS\\5.5.1/CMSIS/DSP/Include" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\variants\\Generic_F411Cx" "-IC:\\Users\\Herman\\Documents\\Arduino\\libraries\\RTOSTicker\\src" "-IC:\\Users\\Herman\\Documents\\Arduino\\libraries\\STM32FreeRTOS\\src" "-IC:\\Users\\Herman\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\libraries\\SrcWrapper\\src" "C:\\Users\\Herman\\Documents\\Arduino\\libraries\\RTOSTicker\\src\\RTOSTicker.cpp" -o "C:\\Users\\Herman\\AppData\\Local\\Temp\\arduino_build_22616\\libraries\\RTOSTicker\\RTOSTicker.cpp.o"
In file included from c:\users\herman\appdata\local\arduino15\packages\stm32\tools\xpack-arm-none-eabi-gcc\9.2.1-1.1\arm-none-eabi\include\c++\9.2.1\bits\stl_algo.h:60,
                 from c:\users\herman\appdata\local\arduino15\packages\stm32\tools\xpack-arm-none-eabi-gcc\9.2.1-1.1\arm-none-eabi\include\c++\9.2.1\algorithm:62,
                 from C:\Users\Herman\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/wiring_constants.h:26,
                 from C:\Users\Herman\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/wiring.h:34,
                 from C:\Users\Herman\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/Arduino.h:36,
                 from C:\Users\Herman\Documents\Arduino\libraries\STM32FreeRTOS\src/portmacro.h:9,
                 from C:\Users\Herman\Documents\Arduino\libraries\STM32FreeRTOS\src/FreeRTOS/Source/include/portable.h:52,
                 from C:\Users\Herman\Documents\Arduino\libraries\STM32FreeRTOS\src/FreeRTOS/Source/include/FreeRTOS.h:62,
                 from C:\Users\Herman\Documents\Arduino\libraries\STM32FreeRTOS\src/FreeRTOS.h:9,
                 from C:\Users\Herman\Documents\Arduino\libraries\STM32FreeRTOS\src/STM32FreeRTOS.h:9,
                 from C:\Users\Herman\Documents\Arduino\libraries\RTOSTicker\src\RTOSTicker.h:31,
                 from C:\Users\Herman\Documents\Arduino\libraries\RTOSTicker\src\RTOSTicker.cpp:25:
c:\users\herman\appdata\local\arduino15\packages\stm32\tools\xpack-arm-none-eabi-gcc\9.2.1-1.1\arm-none-eabi\include\c++\9.2.1\bits\algorithmfwd.h:195:3: error: template with C linkage
  195 |   template<typename _IIter, typename _Predicate>
      |   ^~~~~~~~
In file included from C:\Users\Herman\Documents\Arduino\libraries\STM32FreeRTOS\src/FreeRTOS.h:9,
                 from C:\Users\Herman\Documents\Arduino\libraries\STM32FreeRTOS\src/STM32FreeRTOS.h:9,
                 from C:\Users\Herman\Documents\Arduino\libraries\RTOSTicker\src\RTOSTicker.h:31,
                 from C:\Users\Herman\Documents\Arduino\libraries\RTOSTicker\src\RTOSTicker.cpp:25:
C:\Users\Herman\Documents\Arduino\libraries\STM32FreeRTOS\src/FreeRTOS/Source/include/FreeRTOS.h:52:1: note: 'extern "C"' linkage started here
   52 | extern "C" {
      | ^~~~~~~~~~
```

It compiles OK with : 
```
#include <functional>
#include "Arduino.h"
#include "STM32FreeRTOS.h"
#include "timers.h"
```

fpistm, 2020-10-20:
Hi, 
seems strange as Arduino.h is added on top of the ino sketch when compiling by Arduino.
Could you share a way to reproduce.

hreintke, 2020-10-20:
The problem comes as the libraries are compiled separate from the sketch. -> do not have Arduino.h included.

To reproduce : 
New sketch, include attached library.
No further coding needed.
Compile.
[RTOSTicker.zip](https://github.com/stm32duino/STM32FreeRTOS/files/5407957/RTOSTicker.zip)

fpistm, 2020-10-21:
I've reproduced and found the root cause.
The `FreeRTOS.h` includes the `portable.h`  inside` extern "C" { ... } `.
Then `portable.h` includes `portmacro.h` which include `Arduino.h`.
The `portmacro.h` does not need all Arduino.h definition only some STM32 definition.

```patch
- #include <Arduino.h>
+ #include "stm32_def"
```

hreintke, 2020-10-21:
I applied your proposed update as a  local patch and can confirm it fixes the issue.

Do you want me to close the issue or keep it open until the PR is applied  ?

fpistm, 2020-10-22:
Thanks for the feedback. Issue will be closed by the PR.

PR #34 (2020-10-22): Fix include issue to avoid C linkage issue
Fixes #32
commit a1524e4c (2020-10-22): Fix include issue to avoid C linkage issue

Fixes #32
closed 2020-10-22 by fpistm