stm32duino/STM32FreeRTOS #42: I think it's a bug in freeRTOS config file?

KiraVerSace, 2021-06-25:
Please look the L102
https://github.com/stm32duino/STM32FreeRTOS/blob/master/src/FreeRTOSConfig_Default.h

here is my result I think it should change to  
`#define configTOTAL_HEAP_SIZE             ((size_t)(&_estack - (uint32_t)&_Min_Stack_Size - &_end))`

I use printf to print the value of _Min_Stack_Size it always wrong, but I add the & as &_Min_Stack_Size , it will be the right value that I set in the ld file.

SO I the heap size shoud minus the _Min_Stack_Size set to 0x400?

fpistm, 2021-06-25:
Thank you @KiraVerSace  for pointing this issue 👍

closed 2021-06-25 by fpistm

KiraVerSace, 2021-06-28:
But I still do not know why here should add a &?

fpistm, 2021-06-28:
> 
> 
> But I still do not know why here should add a &?

https://sourceware.org/binutils/docs/ld/Source-Code-Reference.html

```log
Note the use of the ‘&’ operators. These are correct. Alternatively the symbols can be treated as the names of vectors or arrays and then the code will again work as expected: ```

commit 8b196df3 (2021-06-25): fix: cast linker value for configTOTAL_HEAP_SIZE 

Fixes #42