stm32duino/STM32FreeRTOS #44: Trying to use STM32FreeRTOS with Arduino_Core_STM32 V2.1.0

beegee-tokyo, 2021-10-16:
A call for help. I guess I am just missing some setting somewhere.

I am trying to use FreeRTOS with the Arduino_Core_STM32 and the newly added variant **`WL54CCU_WL55CCU_WLE4C(8-B-C)U_WLE5C(8-B-C)U`**

But compilation fails (even with empty sketch) with 
```
C:\Users\RAKWIR~1\AppData\Local\Temp\cci0Ndhb.s: Assembler messages:
C:\Users\RAKWIR~1\AppData\Local\Temp\cci0Ndhb.s:334: Error: selected FPU does not support instruction -- `vstmdbeq r0!,{s16-s31}'
C:\Users\RAKWIR~1\AppData\Local\Temp\cci0Ndhb.s:356: Error: selected FPU does not support instruction -- `vldmiaeq r0!,{s16-s31}'
```

Example sketch:
```c++
#include <STM32FreeRTOS.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

fpistm, 2021-10-16:
Hi @beegee-tokyo 

currently this library does not support the STM32WL. 
I did not have time to add it properly, same for STM32L5 and STM32U5.

beegee-tokyo, 2021-10-16:
Too bad. Anything I can help?

fpistm, 2022-02-23:
Support of CM33 added thanks #48

fpistm, 2022-02-23:
Issue when building `portasm.c.o`
```
C:\Users\***\AppData\Local\Temp\ccMfcvd2.s: Assembler messages:
C:\Users\***\AppData\Local\Temp\ccMfcvd2.s:326: Error: selected FPU does not support instruction -- `vstmdbeq r0!,{s16-s31}'
C:\Users\***\AppData\Local\Temp\ccMfcvd2.s:348: Error: selected FPU does not support instruction -- `vldmiaeq r0!,{s16-s31}'
```

PR #49 (2022-02-25): feat: support STM32WL family
STM32WL has no FPU so fallback to CM3 port as it is done in the STM32CubeWL:

https://github.com/STMicroelectronics/STM32CubeWL/blob/59c5722deaf51724cff8446413a7a76792d8bb81/Projects/NUCLEO-WL55JC/Applications/FreeRTOS/FreeRTOS_ThreadCreation/STM32CubeIDE/.cproject#L48

Fixes #44
commit a08ba4d4 (2022-02-25): feat: support STM32WL family

Fixes #44
closed 2022-02-25 by fpistm