stm32duino/STM32FreeRTOS #50: How to use other timebase for HAL, cause the freeRTOS use the Systick?

KiraVerSace, 2022-04-08:
As we all know it,  When using FreeRTOS, the CubeMX suggests not to use SysTick as timebase for the HAL. I believe this is because there is conflict between the HAL and RTOS use of the SysTick timer?
I found a bug that when I use stm32g070 with STM32FreeRTOS, I use the delay(10) in my setup() function before the osKernalStart,  and my program will always in the delay(10) and never jump out, the HAL_Tick never increase cause they use the same timebase?

How to make the HAL use other timebase as TIM1, TIM2 or other to avoid the conflict?

fpistm, 2022-04-11:
Hi @KiraVerSace 

This limitation is explained in the README.md:
https://github.com/stm32duino/STM32FreeRTOS#limitations
  
> On Cortex-M0 and Cortex-M0+, all IT are disabled between xTaskCreate() and vTaskStartScheduler(). So it is not possible to use IT inbetween, like Serial.print() ... This is the reason why, in example "frLiyLayland", between xTaskCreate() and vTaskStartScheduler(), we use direct printf(), which will access directly USART without interrupt

You are in this case so avoid to use delay at this stage.

KiraVerSace, 2022-04-11:
> Hi @KiraVerSace
> 
> This limitation is explained in the README.md: https://github.com/stm32duino/STM32FreeRTOS#limitations
> 
> > On Cortex-M0 and Cortex-M0+, all IT are disabled between xTaskCreate() and vTaskStartScheduler(). So it is not possible to use IT inbetween, like Serial.print() ... This is the reason why, in example "frLiyLayland", between xTaskCreate() and vTaskStartScheduler(), we use direct printf(), which will access directly USART without interrupt
> 
> You are in this case so avoid to use delay at this stage.
> 
> /cc @ABOSTM if you have any comment on this.

Thank you for your help, you give me the result but do not give me the reason?
I want to know the details, How can I understand the " between xTaskCreate() and vTaskStartScheduler()", I use it before xTaskCreate it would not work, but after the rtos start, it can work ,  I just use it for testing, In the fact, I use vTaskDelay?

Thank you!

KiraVerSace, 2022-04-11:
Sorry  ,May be my reply is not so kind cause I do not good at English,  Thank you sincerely!

ABOSTM, 2022-04-11:
> Thank you for your help, you give me the result but do not give me the reason?
> I want to know the details,

This is the way FreeRTOS works:
See FreeRTOS documentation
https://www.freertos.org/FAQHelp.html

> 4. Interrupts are not executing
> ...
> If a FreeRTOS API function is called before the scheduler has been started then interrupts will deliberately be left disabled, and not re-enable again until the first task starts to execute. This is done to protect the system from crashes caused by interrupts attempting to use FreeRTOS API functions during system initialisation, before the scheduler has been started, and while the scheduler may be in an inconsistent state.
> ...

KiraVerSace, 2022-04-11:
> 

OK, Thank you for your help,.
Now I figure out  the reason why the system will stop in the delay(), there are different results in the different chip [cortex-m4/cortex-mo]although they have the same program. I know the purpose why FreeRTOS do with it.
But I think they are different ways to achieve it between the two chips,  on cortex-m0 all the interrupts will disable until the scheduler is start, but no limitation on the cortex-m3. 
Am I right? HaHa!

ABOSTM, 2022-04-12:
Yes, it work differently depending on cortex:
* Cortex-M0, based on PRIMASK register
* Cortex-M3 or M4, based on BASEPRI register

KiraVerSace, 2022-04-12:
Thank you for your help, I am very happy!

closed 2022-04-12 by KiraVerSace