stm32duino/STM32FreeRTOS #51: Cannot build for STM32L552ZE-Q

asukiaaa, 2022-05-02:
I tried building but failed with the following message.
```
/tmp/ccjgv9nE.s: Assembler messages:
/tmp/ccjgv9nE.s:224: Error: selected processor does not support `vstmdbeq r0!,{s16-s31}' in Thumb mode
/tmp/ccjgv9nE.s:225: Error: instruction not allowed in IT block -- `mrs r2,psplim'
/tmp/ccjgv9nE.s:250: Error: selected processor does not support `vldmiaeq r0!,{s16-s31}' in Thumb mode
/tmp/ccjgv9nE.s:252: Error: instruction not allowed in IT block -- `msr psplim,r2'
*** [.pio/build/nucleo_l552ze_q/lib67e/STM32duino FreeRTOS/portasm.c.o] Error 1
```

It seems similar to https://github.com/stm32duino/STM32FreeRTOS/issues/44 but STM32L5 is supported bu https://github.com/stm32duino/STM32FreeRTOS/commit/adba67836e25becd8ded5bf5a7ba09c2d616769c so I report this.

Is there any idea to avoid this error?

Thank you for sharing the useful library.

fpistm, 2022-05-02:
Hi @asukiaaa ,

L5 support has been added and is available with STM32FreeRTOS 10.3.1.
With Arduino there is no issue. I don't know how PIO handle this but you probably don't have the latest release.
You should request support there as we support only Arduino IDE.

closed 2022-05-02 by fpistm

asukiaaa, 2022-05-03:
I succeeded in building by adding `-mfpu=fpv4-sp-d16 -mfloat-abi=hard` for build commands so I created a PR.
https://github.com/stm32duino/Arduino_Core_STM32/pull/1704