stm32duino/STM32FreeRTOS #52: I find a bug in osSemaphoreCreate

KiraVerSace, 2022-05-25:
https://github.com/stm32duino/STM32FreeRTOS/blob/382c1009adabcf653e861fbcc54bc2cd6bcac486/portable/CMSIS_RTOS/cmsis_os.c#L712

When I try to create a Semaphore with 0, it will in hardfault.

Like 
this->reponseSemID     = osSemaphoreCreate(osSemaphore(responseSem), 0);
cause the  xSemaphoreCreateCounting(count, count); the first count can not be zero.

fpistm, 2022-05-25:
I don't see any issue. Using 0 seems not permitted as stated by the code:
https://github.com/stm32duino/STM32FreeRTOS/blob/12a62916c8e820e2ba7718c6517e9212a0073dc5/src/FreeRTOS/Source/queue.c#L690-L691

Even if you think it is a bug, it has to be reported to FreeRtos as this file is part of it and not changed here.

closed 2022-05-25 by fpistm

KiraVerSace, 2022-05-25:
> I don't see any issue. Using 0 seems not permitted as stated by the code: [...]

thank you for your reply, OK, as your rely, If I want to create a semaphore with 0, how to do it with cmsis-v1, I know it can do in freertos with xSemaphoreCreateCounting(count, 0);

fpistm, 2022-05-25:
I don't know. In all example I never see 0. It is probably documented in Freertos.