stm32duino/STM32FreeRTOS #54: I can not use osSemaphoreRelease(rtcMinuteInterruptSemID) in the RTC alarm IRQ

KiraVerSace, 2022-06-16:
When I use the stm32duino RTC and open the alarm IRQ, I want to release a semaphore in the IRQ, it always failed, then I debug my code, I find thar it was in the assert as blew.

https://github.com/stm32duino/STM32FreeRTOS/blob/382c1009adabcf653e861fbcc54bc2cd6bcac486/portable/MemMang/heap_useNewlib_ST.c#L217

I check my code that the semaphore create sucessfully!

could you help me?

ABOSTM, 2022-06-16:
`osSemaphoreRelease() ` is not the faulty API, as it manages ISR case.
If you can, print your callstack, that would help to identify which API is called in the IRQ handler.

KiraVerSace, 2022-06-17:
Here is my program in the header.h file
```
#include <STM32RTC.h>
extern STM32RTC &rtc;
extern osSemaphoreId rtcMinuteInterruptSemID;
osSemaphoreDef(rtcMinuteInterruptSem);
```
Here is the code in the cpp file
```
/*----------------- RTC ----------------*/
STM32RTC& rtc = STM32RTC::getInstance();
osSemaphoreId rtcMinuteInterruptSemID;

void init(void)
{
	rtcMinuteInterruptSemID = osSemaphoreCreate(osSemaphore(rtcMinuteInterruptSem), 2);
	if (rtcMinuteInterruptSemID == NULL)
	{
		Serial.println("The RTC Semaphore create failed!");
	}
	else
	{
		Serial.printf("The RTC Semaphore create [%d]!\r\n", rtcMinuteInterruptSemID);
	}
}
```

And here is my IRQ Handle
```
void rtcMinuteIRQ(void *parameter)
{
	UNUSED(parameter);

	logInfo("RTC Interrput happend!-----------");

	// osSemaphoreRelease(rtcMinuteInterruptSemID);

	static uint8_t lastDay = rtc.getDay();

	uint8_t minutes = rtc.getMinutes();
	minutes = 1 + minutes;
	if(minutes == 60)
	{
		minutes = 0;
	}

	rtc.setAlarmMinutes(minutes);
	rtc.setAlarmSeconds(0);
	rtc.enableAlarm(rtc.MATCH_MMSS);

	if(rtc.getDay() != lastDay)
	{
		lastDay = rtc.getDay();
	}

	logInfo("RTC Alarm at[%04d-%02d-%02d %02d:%02d], next Interrupt Minutes at [%02d]",
				(rtc.getYear()+2000), rtc.getMonth(), rtc.getDay(),
				rtc.getHours(), rtc.getMinutes(), minutes);
}
```

if I commment this line `osSemaphoreRelease(rtcMinuteInterruptSemID);` the system runs well ,
[image]

if I uncomment this line, the system will print `I(51935483) rtcMinuteIRQ: RTC Interrput happend!-----------`
and enter into the assert in [heap_useNewlib_ST.c](https://github.com/stm32duino/STM32FreeRTOS/blob/382c1009adabcf653e861fbcc54bc2cd6bcac486/portable/MemMang/heap_useNewlib_ST.c#L217)

KiraVerSace, 2022-06-17:
OK, I debug my code step by step

First run to 	`osSemaphoreRelease(rtcMinuteInterruptSemID);`

then I step in, and run to in queue.c` portASSERT_IF_INTERRUPT_PRIORITY_INVALID();`

and run to port.c `configASSERT( ucCurrentPriority >= ucMaxSysCallPriority );`

and into a loop  that is [STM32FreeRTOS/portable/MemMang/heap_useNewlib_ST.c](https://github.com/stm32duino/STM32FreeRTOS/blob/382c1009adabcf653e861fbcc54bc2cd6bcac486/portable/MemMang/heap_useNewlib_ST.c#L217)

KiraVerSace, 2022-06-17:
I have slove the problem, cause of the #define RTC_IRQ_PRIO     is set to 2, it is not in the freeRTOS control priority, so I set it to 7, between 7 and 15

closed 2023-01-27 by fpistm