stm32duino/STM32FreeRTOS #55: Can not use HID bootloader while using STM32FreeRTOS

PadmaIoT, 2022-06-23:
Hi,
I was using the [HID bootloader](https://github.com/Serasidis/STM32_HID_Bootloader/releases) for Arduino IDE for several months. But when I inserted the FreeRTOS I lost the com port for the blue pill. And look like the program does not work. I mean I tried to use pc13 to blink1 and  PA4 to blink2 program (blink 1 & 2 are for different time delays). How can I solve this?

PadmaIoT, 2022-06-30:
Hi,
I am not able to get any answers from the developer until now. Please would you please see the issue?

fpistm, 2022-06-30:
Hi @PadmaIoT 

Which version of the core you used ?
You should try the 2.3.0, we fixed the system init to ensure all configuration is ok after bootloader exit (jump to application).
If it does not work, I have no more idea.

PadmaIoT, 2022-06-30:
I am using 2.3.0 core.

PadmaIoT, 2022-06-30:
Do you have any support for STM32F103C8T6 chip? Bcz I did not find one. @fpistm

fpistm, 2022-06-30:
Yes it is available:
https://github.com/stm32duino/Arduino_Core_STM32/blob/3ef013a908215586b4b738772ba25be588920fdf/boards.txt#L1517-L1523

fpistm, 2023-05-17:
Tested and not reproduced.

closed 2023-05-17 by fpistm