stm32duino/STM32FreeRTOS #60: error: #5: cannot open source input file "cmsis_gcc.h"

bigmirrorDH, 2023-04-10:
I've just started using mdk5, created a new project and the first example compiles with an error """... \... \Drivers\CMSIS\Include\cmsis_compiler.h(41): error: #5: cannot open source input file "cmsis_gcc.h": No such file or directory" "", but I see the same file in the official routine as in the "cmsis_compiler.h" file I imported
I'm not sure how to solve this error

bigmirrorDH, 2023-04-10:
`#ifndef __CMSIS_COMPILER_H
#define __CMSIS_COMPILER_H

#include <stdint.h>

/*
 * Arm Compiler 6 (armclang)
 */
#if defined (__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050)
  #include "cmsis_armclang.h"

/*
 * GNU Compiler
 */
#elif defined ( __GNUC__ )
  #include "cmsis_gcc.h"

/*
 * IAR Compiler
 */
#elif defined ( __ICCARM__ )
  #include <cmsis_iccarm.h>
`

fpistm, 2023-04-10:
Which core you used?

closed 2023-04-11 by fpistm