stm32duino/STM32FreeRTOS #61: STM32duino vTaskStartScheduler() problem

emredaglier, 2023-04-28:
I have a STM32F103C86T with PL2303 to communicate over Serial. I have to implement STM32FreeRTOS but I’m having issues after I call vTaskStartScheduler() in my code. Whenever I call vTaskStartScheduler() after creating my tasks, onboard led of the STM blinks faster than I decided and stays ON for about 3 seconds then starts to blink again. While this happening, my tasks aren’t working at all.

I was trying to test the original example blink code right from GitHub but I couldn’t find a proper solution. Also, even I remove all the led codes from the example (including pinMode()), led still blinks.

`/*
 * Based on Blink_AnalogRead example from: https://github.com/feilipu/Arduino_FreeRTOS_Library
 * Modified by: Frederic Pillon <frederic.pillon (at) st.com>
 */
#include <STM32FreeRTOS.h>

// define two tasks for Blink & AnalogRead
void TaskBlink( void *pvParameters );
void TaskAnalogRead( void *pvParameters );

// the setup function runs once when you press reset or power the board
void setup() {

  // initialize serial communication at 9600 bits per second:
  Serial.begin(9600);

  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB, on LEONARDO, MICRO, YUN, and other 32u4 based boards.
  }

  // Now set up two tasks to run independently.
  xTaskCreate(
    TaskBlink
    ,  (const portCHAR *)"Blink"   // A name just for humans
    ,  128  // This stack size can be checked & adjusted by reading the Stack Highwater
    ,  NULL
    ,  2  // Priority, with 3 (configMAX_PRIORITIES - 1) being the highest, and 0 being the lowest.
    ,  NULL );

  xTaskCreate(
    TaskAnalogRead
    ,  (const portCHAR *) "AnalogRead"
    ,  128  // Stack size
    ,  NULL
    ,  1  // Priority
    ,  NULL );

  // start scheduler
  vTaskStartScheduler();
  Serial.println("Insufficient RAM");
  while(1);
}

void loop()
{
  // Empty. Things are done in Tasks.
}

/*--------------------------------------------------*/
/*---------------------- Tasks ---------------------*/
/*--------------------------------------------------*/

void TaskBlink(void *pvParameters)  // This is a task.
{
  (void) pvParameters;

/*
  Blink
  Turns on an LED on for one second, then off for one second, repeatedly.
  Most Arduinos have an on-board LED you can control. On the UNO, LEONARDO, MEGA, and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN takes care
  of use the correct LED pin whatever is the board used.
  The MICRO does not have a LED_BUILTIN available. For the MICRO board please substitute
  the LED_BUILTIN definition with either LED_BUILTIN_RX or LED_BUILTIN_TX.
  e.g. pinMode(LED_BUILTIN_RX, OUTPUT); etc.
  If you want to know what pin the on-board LED is connected to on your Arduino model, check
  the Technical Specs of your board  at https://www.arduino.cc/en/Main/Products
  This example code is in the public domain.
  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
*/

  // initialize digital LED_BUILTIN on pin 13 as an output.
  pinMode(LED_BUILTIN, OUTPUT);

  for (;;) // A Task shall never return or exit.
  {
    digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
    vTaskDelay( 1000 / portTICK_PERIOD_MS ); // wait for one second
    digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
    vTaskDelay( 1000 / portTICK_PERIOD_MS ); // wait for one second
  }
}

void TaskAnalogRead(void *pvParameters)  // This is a task.
{
  (void) pvParameters;

/*
  AnalogReadSerial
  Reads an analog input on pin 0, prints the result to the serial monitor.
  Graphical representation is available using serial plotter (Tools > Serial Plotter menu)
  Attach the center pin of a potentiometer to pin A0, and the outside pins to +5V and ground.
  This example code is in the public domain.
*/

  for (;;)
  {
    // read the input on analog pin 0:
    int sensorValue = analogRead(A0);
    // print out the value you read:
    Serial.println(sensorValue);
    vTaskDelay(1);  // one tick delay (15ms) in between reads for stability
  }
}`

I’d like to hear some ideas about it.
Thank you in advance,
Emre

emredaglier, 2023-04-28:
Arduino 1.8.15
STM32FreeRTOS v10.3.1

Board "Generic STM32F1 Series" and part number is "BluePill F103CB (or C8 with 128k)"
U(S)ART Support "Enabled (generic 'Serial')"
USB Support "None"
Upload Method "STM32CubeProgrammer (Serial)"

Upload log here:
```

Serial Port ttyUSB1 is successfully opened.
Port configuration: parity = even, baudrate = 115200, data-bit = 8,
                     stop-bit = 1,0, flow-control = off
Activating device: OK
Board       : --
Chip ID: 0x414 
BootLoader protocol version: 3.0
Device name : STM32F101/F103 High-density
Flash size  : 512 KBytes (default)
Device type : MCU
Revision ID : --  
Device CPU  : Cortex-M3

Memory Programming ...
Opening and parsing file: Blink_AnalogRead.ino.bin
  File          : Blink_AnalogRead.ino.bin
  Size          : 17,98 KB 
  Address       : 0x08000000 

Erasing memory corresponding to segment 0:
Erasing internal memory sectors [0 8]
Download in Progress:

File download complete
Time elapsed during download operation: 00:00:02.104

RUNNING Program ... 
  Address:      : 0x8000000
Start operation achieved successfully
```

Also I had no problem with MapleFreeRTOS821 which is Ardunio_STM32 but can't seem to implement it with STM32dunio. I'm new around here

emredaglier, 2023-04-28:
Here's my connection with PL2303. I had no problem with communicating.

[image]

fpistm, 2023-05-17:
Closed as no answers since 2 weeks and not reproduced.

closed 2023-05-17 by fpistm