stm32duino/STM32FreeRTOS #66: Is it a good practice to change HAL time base to timer and free SysTick for FreeRTOS?

jancotipsarevic, 2023-07-16:
Hello all, I am starting playing with stm32duino + freertos + cmake + CLION and everything is working fine but I have doubt.
In my previous work with stm32cubeide + stm32cubemx + freeRTOS it was recomended to change HAL time base to any available timer and free the Systick for FreeRTOS, I have tried to do so but  it looks no ovbious to redefine HAL_TIM_PeriodElapsedCallback function.

It has sense what I am trying to do, is it possible?
I will appreciate any hint on that matter.
Thank you very much.

fpistm, 2023-07-17:
Hi @jancotipsarevic,
by default HT is used by Arduinon API, which explains why you can't redefine the `HAL_TIM_PeriodElapsedCallback`.
To be able to redefine it, you have to disable the HT support for the Arduino API as explained here https://github.com/stm32duino/Arduino_Core_STM32/wiki/HAL-configuration#hal-module-only
So defining `HAL_TIM_MODULE_ONLY` will allow to redefine `HAL_TIM_PeriodElapsedCallback`, pay attention to add `extern "C"` if you redefine it in .ino or .cpp.

About the "sense", I'm not an expert of FreeRTOS. Here, it is just a port as an Arduino library format of the STM32 FreeRTOS version.
All is explained here, the `SysTick_Handler` is used to called the `osSystickHandler`
https://github.com/stm32duino/STM32FreeRTOS/blob/f3abddcd420bd92a862b35877181392bd2647b2a/src/FreeRTOSConfig_Default.h#L218-L224

https://github.com/stm32duino/Arduino_Core_STM32/blob/2e754893e1c5d43096558e8e973ccf5d741a28b1/libraries/SrcWrapper/src/stm32/clock.c#L59-L70

Anyway, you are free to override all of this as it is purely configuration.

jancotipsarevic, 2023-07-17:
Excellent answer you put me on track, thank you very much.

fpistm, 2023-07-17:
Welcome. I close it as considered answered.

closed 2023-07-17 by fpistm