stm32duino/STM32FreeRTOS #69: Documentation suggestion for configCHECK_FOR_STACK_OVERFLOW and for LED_BUILTIN

wystewart, 2024-04-03:
The FreeRTOS web site says that one must implement implement function void vApplicationStackOverflowHook when enabling configCHECK_FOR_STACK_OVERFLOW .  This is not the case for STM32FreeRTOS as it is defined in STM32FreeRTOS.c and there is no option to override.  Suggest to document this and to clarify that the user must ensure that LED_BUILTIN is defined in their variant.h for their board as vApplicationStackOverflowHook will hang (while 1) when stack overflow detected and flash the LED.

Perhaps vApplicationStackOverflowHook and other error functions could be defined as weak in STM32FreeRTOS.c so that the user can override.

fpistm, 2024-04-03:
Hi @wystewart 
Right it was defined by the library but it can't be defined as weak as it already exist one weaked function.
Maybe an other weak function can be added and called by vApplicationStackOverflowHook like this user can redefine it.

Feel free to submit a PR to document or adding extra function or definition.

closed 2024-07-25 by fpistm

fpistm, 2024-07-25:
Hi @wystewart 
Now by default there is no function.

commit d05df76c (2024-07-25): feat: provide default functions only if requested

configUSE_MALLOC_FAILED_HOOK and configCHECK_FOR_STACK_OVERFLOW
allow to define specific function to call when a malloc failed
or a stack overflow occured.
Respectively, vApplicationIdleHook and vApplicationStackOverflowHook.

Previous implementation provide them unconditionally preventing users
to define their own implementation. Now extra config is required to
have the default implementation:

/* Set to 1 to use default blink hook if configUSE_MALLOC_FAILED_HOOK is 1 */
#ifndef configUSE_MALLOC_FAILED_HOOK_BLINK
  #define configUSE_MALLOC_FA [... cut]