stm32duino/STM32FreeRTOS #72: Installing FreeRTOS causes MCU to stop responding

caiotbc, 2025-03-18:
I'm having issue trying to use FreeRTOS on an STM32F103C8 MCU. I don't have any previous experience with FreeRTOS, but simply including the FreeRTOS library on the platformio libdeps causes the board to stop working. Can't get even an LED to blink, even if I don't include the STM32FreeRTOS, just installing the library on platformio is enough to make the board refuse to work. Using the STLink V2 debugger, I get a failure on line 50 of the /home/caio/.platformio/packages/framework-arduinoststm32/cores/arduino/main.cpp, which calls the initVariant();

Removing the library makes everything work again. I've tried several versions, including the latest 10.3.2, with the same result

Here's my INI:
`[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
upload_protocol = stlink
debug_tool = stlink
lib_deps = 
	SoftwareSerial
	stm32duino/STM32duino FreeRTOS@^10.2.1
`

fpistm, 2025-03-19:
Duplicateur of #58.
Particularly this comment 
https://github.com/stm32duino/STM32FreeRTOS/issues/58#issuecomment-1482421973

closed 2025-03-19 by fpistm

caiotbc, 2025-03-19:
Sorry for the duplicate, I assumed it to be related to USB and did not think it would apply to my case, as I am not using any USB functions. Working fine now, many thanks.